markondej/fm_transmitter #163: Increase transmitting power

moom0o, 2022-02-08:
Is it possible to increase transmitting power by editing the code? I have a high speed power adapter for the PI, and would like to try with more power, or is it already the highest it can go?

creeper6530, 2024-01-07:
TL;DR: No, you can't. Also, you don't want to, since it would make it more probable for you to get arrested.

No, you can't increase the transmit power on the Pi alone. Since the Pi has quite low maximum output power per pin, there isn't much one can do inside the code.

You probably could try using an optimized antenna for greater reach, but note that that increases the chance that you'll disrupt someone else. Also, square waves that the Pi transmits have a crapton of harmonics, therefore you will disrupt people outside the FM band as well, which is much larger problem, as those bands might be better monitored and more important, such as military or aviation bands.

But if you want really far reach and do acknowledge that you WILL get into huge trouble, you could maybe use an external booster. Just ensure that you have a good lawyer.

umar4011, 2024-02-28:
Hello can you please make it compatible for raspberry Pi 5

closed 2026-02-06 by markondej