markondej/fm_transmitter #166: Not Working on RPi4

AgentLoneStar007, 2022-02-11:
Hi. Here's my issue. I downloaded and compiled the library just as needed. I added the support for Raspberry Pi 4. I'm broadcasting below 93 MHz. But the song is not playing. When I run the command my FM transmitter explodes in static, but I can't even make out a song. I'm using GPIO pin 40, so what's the issue?

Also, do I need to strip the entire wire, or just the end, or what? I'm using a wire [from this package](https://www.amazon.com/EDGELEC-Breadboard-Optional-Assorted-Multicolored/dp/B07GD2SW11/ref=sr_1_5?crid=JBHC35S5V9MY&keywords=25%22%2Bsingle%2Bwire%2Bfor%2Braspberry%2Bpi&qid=1644548380&sprefix=25%2Bsingle%2Bwire%2Bfor%2Braspberry%2Bpi%2Caps%2C133&sr=8-5&th=1).

CookieGMVN, 2022-04-04:
Which OS you using for your Raspberry Pi 4?

AgentLoneStar007, 2022-04-04:
Raspberry Pi OS(64-bit).

CookieGMVN, 2022-04-04:
Well, I tried to run this software on Raspberry Pi OS 64-bit and it can't work, maybe by some magic errors like in-compatible. Also I have tested and to make this work, you need to install Raspberry Pi OS (32-bit) and not Lite version (I mean the recommended version by Raspberry Pi Imager) then try again to install. If you need better FM Transmitter, you can try Raspberry Pi Zero instead of Raspberry Pi 4, the Pi Zero will transmit the FM waves farther Pi 4.

CookieGMVN, 2022-04-04:
[image]
Choose the first option

AgentLoneStar007, 2022-04-04:
K. Worst-case scenario I have a Raspberry Pi 3 B just sitting, so maybe I can use it.

AgentLoneStar007, 2022-04-04:
Still not working. It's definitely broadcasting because the radio explodes in static when I run the command, but there's no perceivable music. What am I doing wrong? Do I need to strip the wire, or can I leave the casing on it?

AgentLoneStar007, 2022-04-05:
I tried it on my Raspberry Pi 3b+. I can hear the music now, but it's extremely slow. Why would it slow down the track so much?

CookieGMVN, 2022-04-05:
I don't know why it slow down the track because my Pi 4 is working very well and the music is normal but that is super weak.

BobHasNoSoul, 2022-04-16:
can confirm the pi4 is very awkward at the start using those cables if you have a broadcast but you are unable to hear the music pull the jumper lead off and solder directly too the gpio pin a different wire (literally any metal wire i use a 1.5mm copper neutral wire that is around 30ish cm in length.. i was not very accurate at measuring) i know it sounds weird to solder directly to the pin but works really well.. i have just done this and it amazed me how much power these throw out just by changing the way the wire for the antenna was.

AgentLoneStar007, 2022-04-18:
I sadly am not able to do that for three reasons: one, I do not own a soldering iron as of yet. I hope to get one on my birthday. Two is I don't have a bare wire; only insulated wires and they are pretty much impossible to strip. Three is I only have one Raspberry Pi 4, and due to the massive price tag on them, I'd prefer not to ruin the only one I own. I guess I'll mark this as closed, with the reason being I chose a trash wire. Thanks for all the help, though!

closed 2022-04-18 by AgentLoneStar007