markondej/fm_transmitter #168: make: *** [makefile:19: transmitter.o] Error 1

AlienX51, 2022-02-25:
I'm trying to install fm_trasmitter software for my raspberry pi 3, I'm using kali as the OS (since it has a little bit of everything already in the OS and don't need to install majority of the stuff ), I'm trying to compile the code using the line sudo make. and this is what I get:

NOTE: I have updated everything

g++ -Wall -O3 -std=c++11 -DVERSION="0.9.6" -DEXECUTABLE="fm_transmitter"

g++ -Wall -O3 -std=c++11 -c mailbox.c

g++ -Wall -O3 -std=c++11 -c wave_reader.cpp

g++ -Wall -O3 -std=c++11 -fno-strict-aliasing -I/opt/vc/include -c transmitte

transmitter.cpp:36:10: fatal error: bcm_host.h: No such file or directory

36 | #include <bcm_host.h>

| ^~~~~~~~~~~~

compilation terminated.

make: *** [makefile:19: transmitter.o] Error 1

AND when i install dependency this is the error i get:

apt-get install libraspberrypi-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libraspberrypi-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libraspberrypi-dev' has no installation candidate

what can it be..????

AlienX51, 2022-03-02:
i use debian OS and it works. looks like kali linux its a no no for this project

closed 2022-03-02 by AlienX51

harifhaqim, 2024-03-29:
i might be late but fyi,you can just install the libraspberrypi-dev manually via dpkg

harifhaqim, 2024-03-29:
> > i might be late but fyi,you can just install the libraspberrypi-dev manually via dpkg
> 
> Nah, he'd need the `.deb` for manual dpkg install. The problem is with the Apt repos, I think.

yeah, that's what i meant, you'll need dpkg for .deb install... it's a second option if the library or package isn't on the distro repo