markondej/fm_transmitter #169: auido other than demo file plays in slow mo

crack-coder, 2022-04-02:
I was adding files to the folder to play and when I unplugged the pi I was using to move my setup, I started the pi back up and when I turned on the radio to hear the music I put on it was in slow mo. I tried all of the music I put on and I get the same result. I even dragged the wavs over to my computer and they would play normally, and when I played the demo file it would play normally. I didn't do anything new and the ran the same command as before and now the files don't play at full speed.

JadenJSJ, 2022-11-16:
@crack-coder and @mouseroot
I think i found a temporary fix, although kinda hacky it works.
It works by increasing the pitch and increasing the speed of the music.
I also found that using 44100 sample rate yields better quality, so I used it here

`ffmpeg -i 米津玄師.m4a -ac 1 -af "asetrate=44100*(44100/22050),aresample=44100" -f wav -bitexact -acodec pcm_s16le - | sudo ./fm_transmitter -f 105.2 -b 230 -`

markondej, 2022-11-17:
this problem occur mainly on RPi 4, as some other hardware is messing PWM clock and DMA is triggered slower than expected. I found this may be related to ARM cores frequency governor, and by disabling it problem can be solved. Please try typing:
```
echo "powersave"| sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
```

JadenJSJ, 2022-11-22:
@markondej I am using a Raspberry Pi 3B+ and I have the issue

> Please try typing:
> 
> '''echo "powersave"| sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor'''

Dosen't work for me, if I use
'-d 255'
It will say bus error, and exit.

anattolia, 2024-03-15:
I had the same problem and I guess it's a sample rate frequency issue, although I don't know exactly why this happens.

My solution: to duplicate the sampleRate in the `PMWController `class (`transmitter.cpp` file) when you set the parameters of the `ClockDevice`.
As you can see, i literally just multiplied the value (sampleRate) by 2:

[image]