markondej/fm_transmitter #170: Antenna Compatibility

justinkopf, 2022-04-17:
How can I use the fm transmitter with antenna

mouseroot, 2022-04-22:
Unless you go out and buy some kind of jumper cable, you will have to find some kind of spare cable that you can attach to the GPIO4 Pin, In all reality, the antenna isn't really needed and is a bit safer as the distance won't be more than a few feet, attaching an antenna is kinda like stepping outside the legal bounds because even a small 2-inch wire can generate a range of around 50~ish feet. which usually steps outside the legal FCC regulations, however broadcasting under ~100 feet on not-used frequencies is...grey area, generally, you don't want to broadcast outside your house as it can cause all kinds of interference.

As a first test, dont use an antenna, the second test look for a jumper cable and use a small wire to test distance, careful to keep in mind the countries FCC regulations.

justinkopf, 2022-04-22:
Yes, I do have a license for radio operation

creeper6530, 2024-01-07:
> Yes, I do have a license for radio operation

Are you sure that the license covers the band you intend to transmit in? AFAIK the FM licenses cost a fortune and if you have that kind of money, you also have the money for a proper long-distance transmitter