markondej/fm_transmitter #174: Error playing from microphone

jimi-boy, 2022-10-26:
I try to use microphone as input device, I get an error
arecord: main:828: audio open error: No such file or directory

WhatWouldJoeDo, 2022-11-03:
Try to use `arecord -L` and choose the mic you've attached. Example below: 
```
pi@raspberrypi:~/fm_transmitter $ arecord -L
...
hw:CARD=Microphone,DEV=0
    Logitech USB Microphone, USB Audio
    Direct hardware device without any conversions
...
```
then use the word after CARD. In this case it is Microphone. That worked for me.

```
arecord -D hw:Microphone -c 0 -d 0 -r 22050 -f S16_LE | sudo ./fm_transmitter -f 104.4 -
```

dandre15, 2023-06-23:
Hi guys can someone please help me im trying to play the audio from a url instead of a local file but need someone to help thanks