markondej/fm_transmitter #176: pls & m3u files support

Bananonz, 2022-11-11:
It would be good idea to be able to play m3u or pls file directly so it's possible to wrapp a web radio in local

creeper6530, 2024-01-07:
If you can find a way to receive a web radio and pipe the output into STDIO, then it should be possible by chaining pipes. Maybe headless VLC, MPV or something like that.

Something like this:
`receiver -i playlist.pls --stdout | ffmpeg -i pipe:0 --convert-to-wav pipe:1 | sudo ./fm_transmitter -f 96.9 -`
Please note that this is only for illustrative purposes.

Bananonz, 2024-01-07:
I'll try it out in some weeks because the micro hdmi cable of the raspberry broke and now I don't have a screen. Thank you for now! I'll update you soon.

creeper6530, 2024-01-07:
I found out that ffmpeg can have an URL as input, even a live one, so if you can parse/extract the URL from the `.pls` file and feed it as an input into ffmpeg, the pipe chain can look like this:
`ffmpeg -i http://example.com --convert-to-wav pipe:1 | sudo ./fm_transmitter -f 96.9 -`

Haven't tested though, so no idea if it works as I think

Bananonz, 2024-01-07:
I figured how to connect via VNC and I confirm that both ways works. Converting to wav while downloading uses a bit of resources but it's worth it. Thank you for the tip, I think I need to master my knowledge about Linux ;¬)

closed 2024-01-28 by Bananonz