markondej/fm_transmitter #177: Radio Data System not working

Cheerstoast, 2022-11-25:
This is more of a suggestion than an issue, but Radio Data System is not supported, and it would be very nice if you could add the functionality.

If unfamiliar with Radio Data System, it's basically the system that transmits the station name, song genre, description, etc. to the radio receivers.
See wiki:
https://en.wikipedia.org/wiki/Radio_Data_System

I understand this feature might be hard.
Thanks anyway! You deserve another star.

wodowiesel, 2022-11-26:
@Cheerstoast There is actually a version with RDS and other forks: https://github.com/ChristopheJacquet/PiFmRds ;)

Cheerstoast, 2022-11-28:
Hmm. Looks like other people accomplished this!
Developer's enthusiasm never stops, eh?

Anyway, @markondej , keep in mind that RDS would be a valuable feature to add to this project!

closed 2022-11-28 by Cheerstoast