markondej/fm_transmitter #182: How difficult it is to transplant the project to another platform?

Eplankton, 2023-03-15:
I have a Xilinx Zynq and I would like to move this project on it, however, I wonder the requirements of platform-specifically files...

markondej, 2023-03-17:
Keep in mind Zynq is only (but great) ARM-A9 + FPGA platform wihout much application specific components built in. All additional peripherials are expected to be configured/programmmed in PL part and connected to PS with AXI (some basic components like USB, Ethernet, I2C are included in Zynq PS part). This project mainly use General Purpose Clock peripherial in order to generate high-frequency waveforms on RPi's GPIO output. This peripherial maintain PLL configuration, which can be modified 'on the fly' allowing frequency modulation in order to transmitt analog signals. This is spimple and clever approach, but this hardware is RPi-speciffic. As far as I know there are available Xilinx blocks for dynamic clock management, but haven't used them so far, and these require packing into AXI slave for sure. In theory this project may be ported to Zynq platform (as a FPGA/SoC enthusiast I'd love to give it a go but currently don't have any spare time to do it :( )

For SDR applications there are available special Intel/AMD SoC-FGPA families with fast transcivers (check out RFSoC family for Xilinx), but these are not affordable as Zynq-7000 chips and mainly targeted for 5G/military applications (expensive, and require special licenses).

Eplankton, 2023-04-07:
> Keep in mind Zynq is only (but great) ARM-A9 + FPGA platform wihout much application specific components built in. All additional peripherials are expected to be configured/programmmed in PL part [...]

Thanks for your advise, I shall try it by myself.

5ucur, 2023-08-13:
I'll comment here instead of opening a new issue, as it seems closely related.

Do you think it would be feasible, _theoretically_, to port this to PC? Say, broadcasting on a pin on the parallel port (DB-25) (I've seen people use a PC like an Arduino, with eight of the pins on the DB-25 as IO pins).

The main reason I doubt its feasibility is possible high interference from the computer. On the other hand, surely the computer could boost the transmission more than the RPi? Though I believe there are PCI cards out there already, specifically made for this purpose, and USB devices too.

I'm open to the possibility of this not being feasible, and not even actively looking for this program for PC - but I thought I'd ask and see what the dev thinks of the idea.

markondej, 2023-08-14:
> Thanks for your advise, I shall try it by myself.

Just to let you know, clocks signal can be easyly generated/modified using this IP: https://github.com/Digilent/ZYBO/tree/master/Projects/sdsoc/repo/digilent/ip/axi_dynclk_v1_0

> I'll comment here instead of opening a new issue, as it seems closely related.
>
> Do you think it would be feasible, _theoretically_, to port this to PC? Say, broadcasting on a pin on the parallel [...]

Like I wrote before, FM modulation is possible due to use of high frequency clock signal generator which can be driven out via certain Raspeberry Pi's GPIO. If given hardware offers similar functionality (ie. Zynq SoC) then this project might be ported. Transmitting on 100 MHz means the state of GPIO will chage every 10 ns, which is far beyond possibilites of Arudino or even normal PC if done software-way. Propper hardware is must, and I don't realy think DB-25 (an old printer port, I used it as an ATMega programmer) will do.

5ucur, 2023-08-17:
> Like I wrote before, FM modulation is possible due to use of high frequency clock signal generator which is driven out via certain Raspberry Pi's GPIO. If given hardware offers similar functionality (ie. Zynq SoC) then this project might be ported. Transmitting on 100 MHz means the state of GPIO will change every 10 ns, which is far beyond possibilities of Arduino or even normal PC if done software-way. Proper hardware is must, and I don't really think DB-25 (an old printer port, I used it as an ATMega programmer) will do.

Thanks for the input, I'm not much of a hardware person so I wouldn't have known these things before - understandable now!

closed 2026-02-06 by markondej