markondej/fm_transmitter #185: Didn't work at all or rarely

Megamichi123, 2023-05-26:
Hello, I tried the program on a Raspberry Pi 400, unfortunately it only worked very rarely and when I stopped it you heard a slight chin and you heard a click every 2 seconds

I can't speak English, I used google translate
help

stepula, 2023-08-12:
I have to move the mouse to have it to play back

Starman7312, 2025-03-29:
Sounds like the same issue I have, and at first I sat there wiggling the mouse putting an object down to press my keyboard.

I'm using a pi 4B and found that when the cpu idled (I stopped interfacing with the raspberry pi and just had the radio code running) that I got lots of interference noise using either gpio pin 4 or pin 20.

However by running some code that simply loops infinitely and does nothing this solved the issue by preventing the cpu idling. It's an odd solution, but works really well for me.