markondej/fm_transmitter #186: Compile error

fuadxtps23, 2023-06-14:
### g++ -Wall -O3 -std=c++11 -DVERSION=\"0.9.6\" -DEXECUTABLE=\"fm_transmitter\" -c fm_transmitter.cpp
g++ -Wall -O3 -std=c++11 -c mailbox.c
g++ -Wall -O3 -std=c++11 -c wave_reader.cpp
g++ -Wall -O3 -std=c++11 -fno-strict-aliasing -I/opt/vc/include -c transmitter.cpp
transmitter.cpp:36:10: fatal error: bcm_host.h: No such file or directory
   36 | #include <bcm_host.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make: *** [makefile:19: transmitter.o] Error 1###

fuadxtps23, 2023-06-15:
like there is having missing files, but what files' 🤔

markondej, 2023-06-15:
The more important question is: using which OS (distro)? These libraries are normally included in Raspbian OS. I am assuming You are using Raspberry Pi board of any kind.

If You're trying to cross compile the transmitter app try installing these packages before (https://raspberrypi.stackexchange.com/questions/36121/fatal-error-bcm-host-h-no-such-file-or-directory-compilation-terminated):
```
sudo apt-get install libraspberrypi-dev raspberrypi-kernel-headers
```

fuadxtps23, 2023-06-15:
> The more important question is: using which OS (distro)? These libraries are normally included in Raspbian OS. I am assuming You are using Raspberry Pi board of any kind.
>
> If You're trying to cross [...]

im using kali linux

markondej, 2023-06-18:
I will try compiling using kali linux and let you know if this issue can be solved

closed 2026-02-06 by markondej