markondej/fm_transmitter #191: fm_transmitter on Raspberry Pi 5?

jakob-martens, 2023-12-04:
Hi,

it looks like fm_transmitter cannot (yet) run on Raspberry Pi 5. I get the following error when starting the fm_transmitter: Cannot obtain access to peripherals (mmap error)

Any plans to make it compatible with Raspberry Pi 5?

Kind regards,
Jakob

notluca, 2024-02-27:
same issue here :/

umar4011, 2024-02-28:
If you share some more logs and details I can make it compatible for pi 5

notluca, 2024-02-29:
> If you share some more logs and details I can make it compatible for pi 5

When i run `make`, it runs flawlessly. 

But when I run `./fm_transmitter`, this happens.

```Broadcasting at 87.5 MHz with 200 kHz bandwidth
Playing: acoustic_guitar_duet.wav, 22050 Hz, 16 bits, mono
Error: Cannot obtain access to peripherals (mmap error)```

I figured out the GPIO is different for the pi and is done via some RPI1 chip as far as I read. [link](https://www.raspberrypi.com/news/rp1-the-silicon-controlling-raspberry-pi-5-i-o-designed-here-at-raspberry-pi/)

Matthanol, 2024-05-04:
Any progress on this? has anyone found a workaround? can't seem to find anything trough google and the code is a tad to complicated for myself to jump into.

serhadcicekli, 2024-05-28:
Same problem here. Is there a fork that added raspberry pi 5 support?

OlegXio, 2024-06-29:
same issue

xiewoc, 2024-07-06:
perhaps what causes this is because it had not adapt the bcm 2712,the latest version only support pi4b(bcm2711
) (and former)
[image]