markondej/fm_transmitter #194: Setting up a ham radio fox transmitter

bzsasic, 2024-04-11:
I would like to set up a couple ham radio fox transmitters (used for other hams to radio-locate the hidden transmitter). I am a licensed ham, and I have tested transmit on the 2 meter band, and it works great. My question is, I see that the -r function will loop a wave file, but how can I create a pause between transmissions? Basically, I would like to transmit the wave file, then pause transmission for 2-3 minutes, then transmit the same file again. And I would like to continue doing this until the battery runs out. 

is this possible? TIA

creeper6530, 2024-04-12:
If I were you, I'd write a basic Bash script with a loop, in which are the transmission command and `sleep` commands. I'm sure you can google how to do it.

markondej, 2024-04-12:
ie. put this in console:
'''
sudo su [type here your password]
while true; do ./fm_transmitter -f 144.0 my_file && sleep 1; done
'''
Notice there is not "-r" as we are looping the transmitter manually

You can also put second line in .sh file, add execution privilages ("chmod +x my_file.sh") and then run it with sudo

bzsasic, 2024-04-12:
> ie. put this in console:
>
> ```
> sudo su [type here your password]
> while true; do ./fm_transmitter -f 144.0 my_file && sleep 1; done
> ```
>
> Notice there is not "-r" as we are looping the transmitter [...]

Thank you so much for this!

bzsasic, 2024-04-12:
**Follow-up question:** So I played the provided guitar file on 146.5 MHz, and it sounds an electric guitar (totally distorted). Also tried playing my own wave file, and also distorted. I tried different bandwidth settings (amateur band is 20KHz) from 6KHz to 200KHz, and nothing got it sounding better (acceptable). When I transmit on broadcast FM (ex. 89.4 MHz), it sounds perfect. Tried Pi Zero W, Pi Zero 2 W and Pi 3 A+, all with the same result. Any ideas?

creeper6530, 2024-04-12:
I guess it's because of the substantially higher frequency, though I'm not sure at all.

bzsasic, 2024-04-12:
Not sure that's it. The Pi chip is supposed to work up to 230 MHz. I am wondering if there is some kind of low pass/band pass filter in the code that is set for the broadcast FM band (88-108MHz)?

markondej, 2024-04-13:
The problem is in clock resolution, at 146 MHz RPi is not able to modulate sound using whole bandwith linearly. There are 2 or 3 frequencies available in 20 kHz bandwith available. Its not a problem for lower frequencies and 200kHz bw, where the number of useable (for FM sound modulation) frequencies is high. Instead of sine accoustic waves we got squared-like signal like in 8-bit games.

The only possible solution I can see right now is to modulate accoustic signal in delta-sigma high frequency style so it could be pottentially demodulated on reciver side. I have tried this but with poor results.

bzsasic, 2024-04-13:
Thank you for the response. Can you tell me if there is a better frequency I can use between 144.0 and 148.0?