markondej/fm_transmitter #195: Compatibility with Rasberry pi 5

Sima167, 2024-06-13:
Whenever I try to run the fm transmitter I get this error

`Error: Cannot obtain access to peripherals (mmap error)`

i'm not sure if this project is still supported but if you could please fix it on rasberry pi I would love that.

Thank you.

markondej, 2024-09-06:
Since Raspberry Pi 5 in terms of peripherals makes use of completely new architecture (RP1 chip) porting this project to Pi5 will take a lot of effort. Currently I don't have any spare time to make it done :(

BreakyFixy, 2024-09-06:
Thanks for finding the time to reply.
I'm sorry to hear that, it looks like it would be a really fun project.

Skektec, 2024-09-09:
Seem's like a lot of projects aren't compatible with the RPi5, I've ran into compatibility issues quite a lot. Thanks anyway.