markondej/fm_transmitter #198: Any way to use this with mplayer/mpv?

rdap, 2024-11-16:
Using fm_transmitter directly works well enough for testing, but people generally don't keep their media in the form of single .WAV files.

I've tried using snd_aloop and playing to the loopback device, but I just get dead air every time. Has anyone actually gotten this to work with a player besides fm_transmitter itself?

creeper6530, 2024-11-16:
For me piping ffmpeg output works good enough, instead of any sound devices. Plus I'm sure there's some player where it pipes audio files to stdout instead of audio daemon

rdap, 2024-11-16:
I finally got it working with mpv.

Started by using hydranix's solution for piping in audio from a loopback device, and making it a script that can be run as a cron job at bootup:

```
#!/bin/sh

sudo modprobe snd-aloop
arecord -D hw:3,1,0 -c 1 -d 0 -r 22050 -f S16_LE | sudo fm_transmitter -f 85.1 -b 800 - &
```

Then I created a wrapper script called "mpvr" for playing anything I want broadcast over FM (since I don't want the loopback device as the default output for everything):

```
#!/bin/sh

mpv --ao=alsa --audio-device=alsa/dmix:CARD=Loopback,DEV=1 "$@"
```

Works a treat, it just took quite a bit of trial and error to find the correct audio output device for mpv.

closed 2024-11-16 by rdap