markondej/fm_transmitter #201: How can I increase the range of the broadcast?

wozunck12f, 2024-12-11:
Where I live almost no radio stations exist and have no strict laws about occupying broadcast frequencies. What can I do to increase the range? Right now it only works around 2 metres with just a jumper wire.

creeper6530, 2024-12-11:
You could build an illegal amplifier, but I strongly do not recommend it. Even if there's no radio, there's still limit on transmitting power without license.

If you are sure you want to do it, Google something like "FM transmitter amplifier". I can't guide you any further.

BreakyFixy, 2024-12-11:
If you use a longer wire, it can improve the range - we have this project
running on a couple of  Pi Zero Ws  in our office because we all like
different music :-)  With a wire roughly 3 ft (90 cm) long (I cannot recall
exactly how long it is), we have signals that cover the ground floor of our
building and outside the building too :-) We soldered the 22 gauge wire to
a  female jumper, put some heat shrink around the joint, and stretched the
wire out to make the antenna. And because it's not amplified, you shouldn't
run into trouble with any local authorities

wozunck12f, 2024-12-12:
Thank you for the answers