markondej/fm_transmitter #202: Best way to get a clear signal

privatedev11, 2025-01-24:
We're using a pi400 with the aerial on gpio21. We have it on a jumper lead so it doesn't touch any other pins. The signal can be clear for a few seconds then cut to static, and I have to mess around with the lead to maybe get a clear signal. How can I get a constant clear signal? We also have older pis we can test (namely, pi 1, pi 2 and pi 3)

markondej, 2025-01-25:
Honestly, I don't have Pi 400 to reproduce this issue. I will check once again Pi 4 with latest firmware.

privatedev11, 2025-01-25:
They're pretty much the same so whatever works on a 4 will probably work on a 400

privatedev11, 2025-01-27:
We've also just got a pi 5, does this support that board?

Starman7312, 2025-03-29:
I'm using a pi 4B and found that when the cpu idled (I stopped interfacing with the raspberry pi and just had the code running) that I got lots of interference noise using either gpio pin 4 or pin 20.

However by running some code that simply loops infinitely and does nothing this solved the issue by preventing the cpu idling. It's an odd solution, but works really well for me.

creeper6530, 2025-03-29:
@Starman7312 
> I'm using a pi 4B and found that when the cpu idled (I stopped interfacing with the raspberry pi and just had the radio code running) that I got lots of interference noise using either gpio pin 4 or pin 20.

You could also run `echo "powersave" | sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor` (as recommended by the README) to stick the frequency to its minimum. "performance" sticks it to maximum, and "ondemand" is the default when it changes with load.

Starman7312, 2025-03-29:
@creeper6530 

Hi, thanks for the tip.

I have already seen that in the readme, but for my usage scenario it's not ideal - thanks though

creeper6530, 2025-03-29:
@Starman7312 

Forgive my curiosity, but what scenario would make setting the frequency to constant worse than running a looping code that does the same but takes up CPU cycles, hurting performance?

You can change the governor on-the-fly and change it back to ondemand immediately after finishing transmitting, so it can't be for availability, and I'm 75% sure the interference is because of CPU freq changing, not load itself.

Starman7312, 2025-03-29:
@creeper6530 

No problem, please do ask away.

Well my mains concern is that I don't want to run my pi 4B at min clock speed, as I use it for other projects and require higher processing power such as a robotic project etc.

And as I only use the radio occasionally (as a side project) I don't really want to have to keep changing this. I'm aware it's pretty simple to change, but for now I just don't have the need.

Unless I've misunderstood, setting to power saving reduces clock speed to min right? Hence processing power is reduced?

Starman7312, 2025-03-29:
But you make a good point, I may modify my implementation to automatically change the cpu setting, run the code and then change back on exit.