markondej/fm_transmitter #203: Question About Licensing for fm_transmitter Integration

vojtikDortik, 2025-01-25:
Hello,

First of all, I want to say that I absolutely love this project! I’ve developed my own Python code to control the fm_transmitter application on my Raspberry Pi Zero W. Additionally, I created a script that sets up a Wi-Fi access point and a minimalistic Flask web UI. This allows me to connect to the Pi via Wi-Fi from my phone and control the transmitter through the web interface.

Now, I would like to share my project on GitHub, but I’ve encountered a potential licensing issue. Since the fm_transmitter repository doesn’t have a specified license, I’m unsure whether I’m allowed to include or use it in my own GitHub project. As a GitHub beginner, I want to ensure I comply with licensing rules and avoid any violations.

Could you please clarify the licensing status or let me know if I’m allowed to use fm_transmitter in my project? I’d greatly appreciate your guidance.

Thank you so much for your help!
Best regards,
VojtikDortik

markondej, 2025-01-25:
Hello,
the license is included, in old manner, in each cpp file. This is 3-clause BSD license, so feel free to do whatever You want ;).

vojtikDortik, 2025-01-25:
Thank you so much for clarifying and for your kind response!

If you're interested, feel free to check out my GitHub repository: [Raspi-FM](https://github.com/vojtikDortik/raspi-fm/). Please note that it is still a work in progress. I only started uploading the project to GitHub a couple of hours ago, so there is nothing in README.md or some other files.

The project is still in its development phase, but I’m really excited to continue working on it and adding more features in the future. Your project has been a huge inspiration, and I truly appreciate your work!

Best regards,
VojtikDortik

closed 2025-01-25 by vojtikDortik